PiSupply/PiJuice #1017: Communication error

BramWerbrouck, 2023-10-14:
HI, I have - like a lot of other people here - a communication error on my RPI and PiJuice:
[image]

unlike the other posts here, it don't alternate between this communicatin error and 'a normal' screen (it stays on the communication error)

I2U is enabled

what can I do?

tvoverbeek, 2023-10-14:
Which RPi (3, 4, zero, etc.)?
Which OS (Bullseye, Bookworm, other)?
Can you provide the output of running `i2cdetect -y 1` in a terminal?
Maybe obvious: HAT mounted correctly on the 40-pin header?

BramWerbrouck, 2023-10-14:
It is a Raspberry pi 4 with Bullseye.

This is the output of `i2cdetect`:
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --    

The HAT is correctly mounted.

tvoverbeek, 2023-10-14:
This smells like I2C is not enabled.
Check if /dev/i2c-1 exists (`ls -l /dev/i2c*`).
You can enable it via raspi-config -> Interfaces.

BramWerbrouck, 2023-10-15:
This is the output of `ls -l /dev/i`c*:

crw-rw---- 1 root i2c 89, 1 14 okt 22:15 /dev/i2c-1

BramWerbrouck, 2023-10-15:
I changed the dip-switches on the back from
[image]

```
-ON---CTS-
[ON ][OFF] 
--1----2-- 

```
to 
[image]

```
-ON---CTS-
[ON ][ON] 
--1----2-- 

```

and now I have this as output of `i2cdetect -y 1`:

0 1 2 3 4 5 6 7 8 9 a b c d e f
00: -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

tvoverbeek, 2023-10-15:
I suspect your username when executing i2cdetect is not member of the i2c group.
If you run `sudo i2cdetect -y 1` do you get a different output?
The "UU" for address 0x68 means the OS has recognized the simulated RTC in the PiJuice firmware.
Are you on a 32-bit version of Bullseye on your Pi4?

BramWerbrouck, 2023-10-16:
`sudo i2cdetect -y 1` gives me the same result:

0 1 2 3 4 5 6 7 8 9 a b c d e f
00: -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

BramWerbrouck, 2023-10-19:
Hi,

I found a similar problem and this was (then) the solution:

_What did help somehow was to put__
`dtoverlay=i2c-bcm2708`
_into the /boot/config.txt. That uses the older i2c-bcm2708 module instead of the default i2c-bcm2835._

But this has no effect with my RPI. Is there something else I can try?