PiSupply/PiJuice #1024: Start RPI when Powerup on USB

BramWerbrouck, 2023-10-28:
Hi,

Finnaly I got my piciuce up and running. I got a script that runs on a power down (when removing the USB power from tha hat) but I'm looking for the right settings to powerup the RPI when the USB power is connected to the hat, so I don't need to push the onboard button everytime.

tvoverbeek, 2023-10-28:
Set 'Wakeup-on-charge' to a very low tthreshold (e.g. 0 or 1) and check the restore option (this stores the setting in the MCU non-volatile memory). As soon as you apply external power via the PiJuice USB the firmware detects charging and because of the low threshold the 5v to the Pi is turned on and the Pi boots.

BramWerbrouck, 2023-10-29:
I set the Wakeup)on-charge to 10 (after I tried 0 and 1) but I still need to press the button to startup the RPI.. The external power is (dis)connected to the PiJuice hat directly (not to the RPI)

bablokb, 2023-11-11:
I have the same problem. Firmware 1.6

bablokb, 2023-11-11:
Note: when I try this without a battery, it works. I.e. if I plug in USB-power, the system boots. It just does not do this if there is a battery attached.

tvoverbeek, 2023-11-12:
@bablokb What is the output from this script:
```
#!/usr/bin/python3
from pijuice import PiJuice
pj = PiJuice(1,0x14)
print(pj.power.GetWakeUpOnCharge())
```
Does it match with the setting shown in the CLI/GUI?

bablokb, 2023-11-14:
Reading the instructions really helps. Once you tick "restore" it works.

My suggestion is to change the gui. From an UI perspective, there is an "apply" button, and as a normal user I would expect that my settings are applied. "Restore" in contrast sounds like "restore defaults" and not "store into nvram".

You have very powerful hardware, and of course the software cannot be simple and support all options at the same time. But it should be consistent so users are not confused. This should also limit the rate of false issues. But I know that software (especially user-interfaces) is a very tedious business.

tvoverbeek, 2023-11-14:
I agree with the "bad" choice of Restore.
There are more problems with this feature. Querying the wakeip-on-charge threshold always returns the non-volatile value if it is defined.