PiSupply/PiJuice #1025: Problems with PiJuice overheating

sebastianpiccoli, 2023-11-06:
Hi,

We have a Raspberry Pi3B+ 1GB RAM and a PiJuice HAT. The battery seems to constantly overheat to 42/43°C even when there is little processor usage (around 7%).

We tried to set a custom battery profile using pijuice_cli and pijuice_gui but the changes don't get saved and get back to default when we reboot the device.

We upgraded the firmware to the latest (v 1.6) but in the newer version the problem persists and we have another issue: the temperature shown is 24°C even when the battery is still around 42°C (this measured using the J2 pinouts).

So we have basically 3 issues:
- Overheating of the battery
- Battery profile changes not saved
- Wrong temperature reading on pijuice 1.6.

Kind regards,
Sebastian.

tvoverbeek, 2023-11-07:
Are we talking about the PiJuice HAT or the PiJuice Zero?
Only the Zero has a J2 battery connector. No way to measure the battery temperature directly,
unless you process the NTC measurement yourself.
How do you know profile changes are not saved?
You can only save the custom profile and that is stored in non-volatie memory.
This survives reboot since the custom profile is stored in non-volatile memory.
Did you press the Apply button in the GUI after defining the custom profile.
Also if you want to see the contents of the custom profile after selecting it you have to press the Refresh button.

sebastianpiccoli, 2023-11-08:
This is a PiJuice HAT, picture attached:
[image]

This is what we mean by J2 (image taken from [here](https://learn.pi-supply.com/make/pijuice-quick-start-guide-faq/)):
[image]

Yes, of course we pressed appy in the GUI (and in the PiJuice Settings on graphic mode, since this issue is happening on both). When we reboot we see the default profile loaded, even when we have saved this:
[image]

Kind regards,
Sebastian.

sebastianpiccoli, 2024-01-17:
Any updates regarding this?

Regards,
Sebastian.

tvoverbeek, 2024-02-01:
For me changing the battery profile works, also over reboots.

Starting with the default BP7X battery profile:
[image]

Changing a few parameters: Warm and hot temperature and selecting Custom:
[image]

Scrollling down and select Apply Settings:
[image]

Confirmation screen:
[image]

Then reselecting Battery Profile:
[image]

Profile name is now Custom.

Restarting pijuice_cli still shows battery profile Custom (with the changed BP7X values).
Same after reboot.
Same after shutting down, removing the HAT, remounting the HAT and booting.

So I do not understand why your custom profile is not saved.

sebastianpiccoli, 2024-02-02:
We change the profile to custom, apply it. We exit pijuice_cli and run it again. Everything works fine and we see the custom profile being set:
[image]

When we reboot it, the profile has been set to default again:
[image]

One thing worth mentioning is that we cannot run pijuice_cli as root, it errors out when we go to "Battery profile". We can set the values only running pijuice_cli with user pi:
[image]

sebastianpiccoli, 2024-02-02:
Hi,

We have finally found the root of our problem. We found a script called "wakeup" which was resetting the battery profile on startup. We can now close this issue. Thanks for your support.

Sebastian.

closed 2024-02-02 by sebastianpiccoli