PiSupply/PiJuice #1026: PiJuice won't communcicate with GPI header - failure to communicate issue.

nlandas, 2023-11-07:
I've tried all the suggestions on here to try to get the pijuice HAT to communicate with the pi through the GPIO connector. It will rarely show up on 14 and 68, I was able to get it to connect long enough to reflash the latest firmware. I've tried all the suggested boot.cfg parameters without luck to try and address any conflicts. 

I've tried a fresh raspberry pi OS image, since I thought it might be the OS that came with moode audio causing the problems. Nothing that I do will allow me to get the pijuice to talk with the configuration software. I can't return the unit to the shop that I bought it from until "........we require authorization from PiJuice for an exchange or return"

I've spent over 3 hours fiddling to try and get this to communicate based on all of the suggestions in the project forums.

tvoverbeek, 2023-11-07:
Moode audio seems to use I2C to control some display.
Your best bet is to check with the Moode developers if this can interfere with other I2C devices like the PiJuice.

nlandas, 2023-11-07:
Thanks for the reply. 

I can certainly contact moode audio to see if they can modify their product but I also tried a stock Raspberry Pi OS install and the pijuice would not communicate with it either. The i2c output was slightly different without either displaying nothing as in all "--" or "UU" where 68 would display if connected.  

So the software still could not communicate with the pijuice HAT. I understand these are hobbyist type projects and appreciate the community builds, Shouldn't the pijuice work out the box with 64-bit raspberry pi OS. I even tried lite just to see if it had less packages but didn't figure the desktop was likely to be the problem.

tvoverbeek, 2023-11-07:
'UU' on position 68 means the OS has loaded the ds1307 module for the (simulated) RTC in the PiJuice firmware. This means I2C is working. Normally i2cdetect should identify addresses 14 (the PiJuice firmware) and 68 or 'UU' for the RTC.
There is a problem when you are using a 32-bit OS on Pi3 and later. Lately the Pi3 and later when using a 32-bit OS use a 64-bit linux kernel. With a normal installation pijuice_cli and pijuice_gui fail to run because they try to run a 64-bit app while the OS is expecting a 32-bit app.
Could that be your problem?

nlandas, 2023-11-07:
Thanks again. i2cdetect would not show the firmware 14 the majority of the time and would occasionally say 68 or UU in the RTC spot. Most of the time it returned all "--" in every position. I did manage to get the 14 firmware up long enough for a successful flash. Based on what you just conveyed the RTC shows up better on the stock Rasp OS but not the firmware. 

The Raspberry PI OS that I installed was 64-bit. I seldom use 32-but unless called for by some older library. The behavior of this issue is largely like many other posts and resolution seems to be very mixed, with only a few people reporting that changing boot.cfg parameters helped correct the issue. 

This is a new Pi 4, so based on your details, if I had run a 32-bit OS that could have been the issue. I used the Pi imager to create the 64-bit microSD card.

nlandas, 2023-11-16:
Thanks for the responses, can I please have pishop receive permission to accept a return on this unit? I honestly spent at least 3 hours trying different boot cfg combinations and different fresh OS instances. Still can't get this to communicate over the GPIO header the software just fails to communicate.