PiSupply/PiJuice #1029: Section "Sync RTC time at boot" - legacy instructions

bablokb, 2023-11-11:
With a halfway up to date PiOs, you don't need to update the system clock as described in the software-section. This is now automatic once the kernel detects the rtc. In consequence, you just have to add the overlay regardless of the EEPROM-address (as described in the section above).

As a side note: in the section you recommend updating the rtc in `/etc/rc.local`. Note that `/etc/rc.local` was already legacy in 1983. 50 years ago.... In modern Linux, you use either udev-rules or systemd-services for things like that. `/etc/rc.local` is about the last thing that is executed during boot, but updating the system-time should be done as early as possible.

But wait: there is a section ("Repeaded wakups") where you recommend @reboot in the crontab. Now this is really really legacy :-(

tvoverbeek, 2023-11-11:
Note: the PiJuice RTC is simulated in the firmware. When Pi/PiJuice are both off the RTC is not running, so you canot rely on the PiJuice RTC to have the correct tie at boot.
Basic issue is the OS detecting a DS3239 (clock with alarm)  due to the device tree fragment in the HAT eeprom. 2 captains controlling the RTC does not work.

bablokb, 2023-11-14:
I rechecked and adding the overlay works just fine. The system time will be automatically updated from the RTC and I can still control the alarm and wakeup from the gui. The RTC also does not loose it's time either as long as there is a power source (usb or battery) connected.

tvoverbeek, 2023-11-14:
Closing this. If you do not agree, please reopen it

closed 2023-11-14 by tvoverbeek