PiSupply/PiJuice #1030: installation did not add user pijuice to the group i2c

bablokb, 2023-11-11:
I recently installed pijuice on a Bullseye system. It did not work as expected, the pijuice.service did not start. Tracking down the problem showed that the installation does not add the user pijuice to the i2c-group. Once added manually, the service worked fine.

tvoverbeek, 2023-11-11:
????
The postinst script for pijuice-base creates the pijuice user and adds it to the i2c group.
See lines 3-12 in https://github.com/PiSupply/PiJuice/blob/master/Software/Source/debian-base/postinst

There is a problem with the pijuice-base (and pijuice-gui) install on bullseye when running on a 32-bit user space system. See e,g, #1000. Already fixed in github. Waiting for a Pi5 before updating the packages.

bablokb, 2023-11-14:
This is indeed strange, because the pujuice user was created but not added to the i2c group. I will try again on a clean install. BTW: I am using a 32-bit kernel/OS.

Two things to note about the postinst script (unrelated to this issue):

- it silently assumes that there is a default user with id 1000, which is not the case for me
- python3.9 is hard coded. A better option with less support work would be to query the python3-version dynamically.

tvoverbeek, 2023-11-14:
The postinst is geared toward Raspberry Pi Os. When you use the Raspberry Pi Imager to burn your SD you can specify a different username (which in fact is recommended) than pi. The first user is assigned uid 1000 in Raspberry Pi Os. Hence the  assumption id 1000 is the default user.

It puts the package in /usr/lib/python3.9/dist-packages.
Note there is a symbollic link to /usr/lib/python3/dist-packages (line 30 in postinst).
/usr/lib/python3/dist-packages should be in the PYTHONPATH for any 3.x version.

bablokb, 2023-11-14:
As soon as Debian changes the python version to say 3.11, you have to change your code (line 30 in postinst). Why not use `python3 --version | sed -e 's/Python //' | cut -d'.' -f1-2` to query the version dynamically?

Regarding id 1000: I provision my users differently, so even though I use Raspberry Pi Os I don't have a default user with id 1000. But although this is a wrong assumption I would not recommend to change this, because this is a fair guess for most users.

tvoverbeek, 2023-11-14:
i  have a bookworm (debian 12) pi4 which has python3.11 as default.
The PiJuice install worked without problems on both systems.

On buster I get this:
```
tvo@rpi64-deb11:~ $ python3
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys; print('\n'.join(sys.path))

/usr/lib/python39.zip
/usr/lib/python3.9
/usr/lib/python3.9/lib-dynload
/usr/local/lib/python3.9/dist-packages
/usr/lib/python3/dist-packages    <------
/usr/lib/python3.9/dist-packages
>>> 
```
On bookworm I get:
```
tvo@rpi64-bookworm:~ $ python3
Python 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> print('\n'.join(sys.path))

/usr/lib/python311.zip
/usr/lib/python3.11
/usr/lib/python3.11/lib-dynload
/usr/local/lib/python3.11/dist-packages
/usr/lib/python3/dist-packages     <-------
/usr/lib/python3.11/dist-packages
>>> 
```
Because of the symbolic link in /usr/lib/python/3.x/dist-packages/pijuice.py the module is found in any python3.x version:
```
tvo@rpi64-bookworm:~ $ ls -l /usr/lib/python3/dist-packages/pijuice.py
lrwxrwxrwx 1 root root 43 Oct 19 21:14 /usr/lib/python3/dist-packages/pijuice.py ->
/usr/lib/python3.9/dist-packages/pijuice.py
```
As long as we stay in the python3.x family there is no problem.