PiSupply/PiJuice #1034: PiJuice: global failure, randomly decides to stop working. The battery must be removed/reinserted

KevinJoss, 2023-11-16:
Hello,

I'm opening this issue about the PiJuice UPS module, which I'm using for a project based on a Raspberry Pi4. 
We have already ordered more than a dozen of this modules and everything has always worked well. 
That was until last month, when we had a problem with an UPS module. In fact, I got the message "COMMUNICATION ERROR" in the PiJuice Settings application on the RPI4. Suspecting a fault with the module, we ordered a new one, but alas, I'm having exactly the same problem. 

Digging around, I found that even when using the UPS without any RPI, I also have this problem. When I put the battery in, the LED lights up to indicate the state of charge (red, green depending of the current charge) and then goes out. When I click on the SW1 button, the LED comes back on. However, this only works for about 5 minutes and then nothing works. Nothing happens if I click on the SW1 button again, and there's no life flashing (which normally happens every 3-4 seconds, as I've seen on modules that work) and I have to remove the battery and put it back in for it to work again (for 5 minutes). 

If I put the UPS on the RPI4, when I manage to get it to communicate (after putting the battery back in), I have communication. But if I turn it off (by removing the power supply), the RPI will gently shut down (the goal of this UPS). Then if I put the power back on and the RPI4 restarts, I won't have any more communication. This is very strange. It's like it's "locking up" and removing/reinstalling the battery allows it to "re-initialise" itself. 

Have you ever had or heard of anything similar? Once again, I've had no problems whatsoever with a dozen or so applications and I haven't changed my hardware. I have the same RPI4. The batteries are all up to 80% charged.
The PCB version of the 2 problematic modules is V2.2.3, but I also have this version on other modules that work.

For the automatic shutdown, when the power is cut, I count 5 seconds in my Pyhton application and if I still have no power, I trigger the shutdown using the following command:
`self.__pijuice.power.SetPowerOff(RPI_UPS_TIMEBEFORECUTOFFBATTERY)`
Then, following this command, I ask for the OS to be shut down.

I look forward to hearing from you.
Best regards

Kevin Joss
Unitechnologies SA

leighghunt, 2023-11-22:
Following this, as I've used about 15 of these in a remote application relatively reliably, and of the last 5 I've deployed with a new batch of PiJuices, 3 are having intermittent unexplained waking issues. All are running the same setup, hardware and firmware, so I don't think it's a software/configuration issue.

All are deployed other end of the country, so inconvenient to diagnose, and this seems to be becoming the common single point of failure for my project.

Symptoms include failure to wake on schedule, failure to wake on SW1 button press, and sudden and continued communications failure of PiJuice board after working reliably for days/weeks.

Mine are all PiJuice Zeroes.

KevinJoss, 2023-11-23:
Hello leighghunt,

Thank you for sharing your experience. Reading what you've said, it sounds a lot like my problem. The difference is that I've been able to try removing and replacing the battery when it stops working, and I've found that it works again for a while after that. Which isn't possible for you, as your modules are on the other side of the country.

And this, despite the fact that in your case it's the PiJuice Zero, I imagine that apart from the shape of the PCB, its contents are relatively the same and that the problem seems to be linked to the hardware.

What have you done in your case?

leighghunt, 2023-11-23:
> Hello leighghunt,
>
> Thank you for sharing your experience. Reading what you've said, it sounds a lot like my problem. The difference is that I've been able to try removing and replacing the battery [...]

I've ended up getting my client, and latterly a support person at my expense, to manually reset them, either by rebooting with SW1, or by completely unplugging everything and resetting - but this has started not working for some PiJuice boards, 3 or 4 now, which I've had to replace. It's ceasing to be a viable option and I'm going to have to look elsewhere for a solution I think.

I suspect you're right about the zero being largely a format difference only.

My project is a set of remote solar powered cameras, for use in originally in long-term time-lapse, and latterly environmental pollution monitoring - the current batch have been running for about 7 months, some flawlessly, others not so much - more info here: https://github.com/venari/timelapse.

pblottiere, 2024-01-23:
Hello everyone,

Unfortunately, I think I have the same issue with a RPI4 :/. "Sometimes", the PiJuice doesn't respond properly and I need to remove/reinsert the battery to make it work again.

In this particular case, I noticed that the LED D1 is strangely blinking (the sequence is not described in the README):

1. led is off for ~2 seconds
2. led is red with LOW intensity for ~1 second
3. led is red with HIGH intensity for ~ 250ms
4. -> 1.

Does someone know what it means?

Thanks for any help :)

leighghunt, 2024-01-23:
I've been doing further troubleshooting of my cameras over the last couple of months, and whilst this may not be relevant if you're using the normal pi Hat setup, one thing that might be a factor in my setup is the wires used to connect the battery to the PiJuice - they're not particularly long or thin, but I've found shortening them to 3-4cm seems to help the failure to wake issue. Seems like there's a larger than expected voltage drop across the cable during Pi startup (the Pi is attempting to start but not completing it). I've still got more testing to do before conclusively determining this is my issue.