PiSupply/PiJuice #1035: Wakeup settings lost when battery is exhausted

hplato, 2023-11-18:
I have added the wakup command to my cron:

@reboot /usr/bin/python3 /home/pi/PiJuice/Software/Test/wakeup_enable.py

I use the rpi to wakeup at 3:00am and to fetch some information to write to an eink display. When the battery dies, I plug it in for a few hours, then press the power-on button, which restarts the PI to pull down the daily information.

I've noticed that often when the battery dies, the wakeup settings are blank. Is there any way to programmatically set these at a reboot so that it can reliably restart on a periodic basis?

hplato, 2023-12-01:
In the pijuice_util there is a --get-alarm that provides some JSON with the alarm details. Looking through the library there is an rtc.setAlarm function that take an object and sets time. My python is pretty rusty, is there a code example of how to use SetAlarm to programatically set the pijuice alarm parameters?

tvoverbeek, 2023-12-03:
Look at how picuije_cli and pijuice_gui do it.
For pijuice_cli look at _set_alarm starting on line 1174  (https://github.com/PiSupply/PiJuice/blob/master/Software/Source/src/pijuice_cli.py)