PiSupply/PiJuice #1040: Pi 5

sevdh, 2023-11-25:
I’m curious to know up to which extend the PiJuice is compatible with the new Raspberry Pi 5.

shawaj, 2023-11-26:
It should work fine with Pi5 but we haven't got one yet to test with.

So will leave this open

sevdh, 2023-11-26:
Good to know! I’ve ordered one but don’t have a delivery date yet…

Two aspects I’m curious about:
- The Pi 5 distinguishes between a 3A and 5A power supply. However the PiJuice delivers less than 3A. What is the possible effect of this?
- The Pi 5 has the option to connect a battery for the onboard RTC. Could This open other possible scenarios for the RTC synchronization between the Pi and the PiJuice?

Thanks for leaving the issue open.

tvoverbeek, 2023-11-27:
I have been using the PiJuice on a Pi 5
It works, but there are problems.
Initiating SYSFUBC_HALT_POWER_OFF via SW1 shustd doen the system, but it imeediately reboots..
Do not know why. Maybe some interaction with the Power Management Chip on the Pi5?
Running the desktop without heavy load works fine when the power is supplied by the PiJuice only.

Regarding the RTC. The system RTC is the one in the Pi 5 Power Management chip.
This is /dev/rtc a symbolic link to /dev/rtc0
The PiJuice RTC is /dev/rtc1 and is independent from the system RTC.
Have not tried to check if the PiJuice RTC picks up the right time form the system (running with
internet, so timesyncd is active)

sevdh, 2023-11-28:
Nice to hear the Pi runs OK on the PiJuice without heavy load. I'm curious what the boundaries are and what happens when exceeding the available power. Will for example the CPU clock speed be limited or could worse things happen?

Do you think it is the SYSFUNC_HALT_POWER_OFF script that would always make the Pi reboot (also when called in situations independent from SW1) or is it SW1 specific?

Thanks for the information on the RTC. I would be happy to hear it when you have more details on the RTC dialog (and also on other interactions between the PiJuice and the Pi5).

Wahaj6Ahmad, 2023-12-12:
Hello all,
I am trying to set up pijuice on the new pi5. I notice the pijuice gui does not open, and neither does the tray option show up.
I installed the pijuice gui through the following command: sudo apt-get install pijuice-gui
I am running the latest 64-bit bookworm OS on the Raspberry Pi 5.
@tvoverbeek You mentioned working with pijuice and pi5. Are you able to install the Pijuice gui without problems?
I also noticed the config file only has this one line:
"system_task":{"enabled": true}}
All other settings are missing. Any idea why?
Lastly, I noticed some of the Python scripts using pijuice are working. I'm curious if I can observe the changes made by these scripts somewhere. Earlier, either the GUI or the config file would reflect the changes set by the scripts. E.g.: I set wake-up alarms through the script but the Pi does not wake up, and I wonder if the alarm is indeed set.

tvoverbeek, 2023-12-12:
Pijuice gui uses the Tk toolkit which is not supported under Wayland.

Start `sudo raspi-config` in a terminal and go to Advanced->Wayland.
Select use openbox and X11. Then pijuice-gui and the tray icon work.
To work under wayland pijuice-gui has to be rewritten using e.g. PyQt5.

tvoverbeek, 2023-12-17:
Correcting my answer above (https://github.com/PiSupply/PiJuice/issues/1040#issuecomment-1828654003)
On the PiJuice Hat General Config tab in the GUI (https://github.com/PiSupply/PiJuice/tree/master/Software#pijuice-hat-general-config-menu)
or the GenerL Tab in the CLI (https://github.com/PiSupply/PiJuice/tree/master/Software#general)
make sure Power regulator mode is set to POWER-SOURCE_DETECTION.
If it is not, the battery will not charge when the Pi5 is powered via the USB-C.
(Had the wrong setting in my comment above)
With the correct setting eveyrhing works as it should on the Pi5 for normal use. For heavy power loads there will be problems, normal desktop use is OK.

Wahaj6Ahmad, 2023-12-22:
Hey, thank you for your comments so far.
I am also curious what 'isFault' = True means when I get the Pijuice status.
`PiJuice Status:: {'data': {'isFault': True, 'isButton': False, 'battery': 'CHARGING_FROM_IN', 'powerInput': 'PRESENT', 'powerInput5vIo': 'NOT_PRESENT'}, 'error': 'NO_ERROR'}
`
Note that the PiJuice Gui shows no fault. Picture attached below.
[image]

InfiniteBSOD, 2024-03-15:
I'm interested if there has been an update to this since I'm considering buying this for use with a RPi 5 w. 8GB of RAM.