PiSupply/PiJuice #1042: PiJuice Hat goes dark

johnbradley, 2023-11-29:
I have a PiJuice Hat that will not stay on for more than a few seconds.
It is connected to a Raspberry Pi 4B.

When the battery is initially connected to the HAT the main led comes on shines a dim green color.
After a few seconds it shows green bright.
After a few more seconds the led goes dark and the PiJuice stops working completely.

Connecting a known good USB cable to the HAT charging port doesn't cause it to come back on.
If I connect power to the Pi it cannot communicate with the HAT.
If I run pijuice_cli and navigate to status it shows `Fault: COMMUNICATION ERROR`.
If I run `i2cdetect -y 1` it shows only `--` for all addresses.

johnbradley, 2023-11-29:
If I quickly press the button to power on the Pi after inserting the battery. The PiJuice Hat will power up the Pi. I can run `pijuice_cli` and see the battery status (it was around 50%). But then after a few seconds it switches to `Fault: COMMUNICATION ERROR`.

johnbradley, 2023-11-29:
I have another PiJuice Hat that works fine staying on as expected.

Martin-HiPi, 2023-12-12:
PiShop.us is awaiting a reply to help our customer, what should we do?

Martin-HiPi, 2023-12-12:
> PiShop.us is awaiting a reply to help our customer, what should we do?

Unit replacement has been approved, Thanks!

closed 2023-12-18 by tvoverbeek