PiSupply/PiJuice #1045: Control GPIO with pi shut down?

quitmeyer, 2023-12-03:
I wrote to the support and they told me to post my questions here:

So our pi controls some relays which are hooked up to bright UV lights (for luring insects). The relays are on a waveshare relay board and just turned on and off with GPIO commands. These lights are on for about 5 hours a night, but the raspberry pi only needs to capture photos about every 5-10 minutes.

I have already tested turning the pi all the way off, and then all the way back on via python scripts and those are working fine. But i was curious if you knew of a way to wake up a pi just enough to let the GPIO turn on the relays, and then turn the pi on the rest of the way to capture the photos when needed? The idea being that over a 5 hour period we don't really need the pi to be running at full capacity for 5 straight hours, but we do need the lights on.

Like is there a way to shut down the pi but leave the pi's gpios in whatever state they were?

tvoverbeek, 2023-12-03:
you asked `Like is there a way to shut down the pi but leave the pi's gpios in whatever state they were?`
Short answer No.
You need a separate power supply for the UV lights which you can switch on/off via GPIO.

tvoverbeek, 2023-12-03:
You need some form of a latching relay. Then it will not change state when the GPIO output goes off at Pi shutdown.

quitmeyer, 2023-12-10:
(Sorry I still want to try to use a regular relay, i understand the latching relay would be the smart way to go, but there seems to be maybe a way to pull off this trick with the pijuice! and I already have to use dozens of these regular relays I am using for a project we already have, and it seems like it might almost be able to work, with maybe some quick trick with the pijuice settings or something!)
My setup:
Pi 4 + Pijuice + waveshare relay hat (https://www.waveshare.com/wiki/RPi_Relay_Board)
The relay hat uses RPI pins 37,38,40, and works totally fine when everything is on. The site says it uses 3.3v 5v and GND from the GPIO

If i have a relay connected to something like a UV light with an external power source it will turn on when the relay connects, and off when the relay doesn't connect. And when I do a full shutdown with the PIjuice and turn off all the power, the relay and UV light of course shut down.

but! if I just tell the raspberry pi something like

sudo shutdown

The GPIO pins still have power and the relay had indicates it still has power.  I checked the 3.3v 5v and GND pins and they all seem to be operational and have power to them.

The really interesting thing to me is that the relays seem to keep the state of how they were when the pi shut down. For instance is Relay 1 and two are off and i measure their continuity with a multimeter between the NO connections they show no connection (as they should). but if Relay 3 is on, the multimeter shows that the NO connections are actually touching! 

Now the confusing thing to me is that the UV light connected to these connections isn't on even though the multimeter says they are connected? Which doesn't make any sense to me, right? isn't the relay just a binary switch that's either not touching or touching? 

Anyway my only thought is that maybe there is some kind of mystical thing happening like the GPIO is still on and the pins have voltage, but they are somehow current limited in this state? and thus maybe the relay switch is being pulled over, but just very slightly? (not saying this makes much sense)

but maybe someone knows something in the pijuice that's like, "oh just enable the special wakeup on boot GPIO pin, and it gives full power to the GPIO in a shutdown state" that would be great! (This is just an example, im not a pro at this, but trying to figure out something potentially cool!)