PiSupply/PiJuice #1048: Wayland not working in Bookworm, X11 does though.

allphasepi, 2023-12-13:
Has anyone got PiGui working in Wayland I had to revert to X11 to get it working. It wouldn't load the preference screen even though there was an icon in the drop down menu. No battery icon either. I installed Bookworm to one of my Pi 4s whch has a PiJuice hat.

tvoverbeek, 2023-12-13:
The tray icon only works when you restart the desktop after changing the setting to OpenBox + X11.
Also the default installation does not detect 32-bit vs 64-bit correctly on a Pi3, Pi4 and Pi5.
See e.g. issue #1000

For me the Gui works fine on a Pi5

tvoverbeek, 2024-01-01:
pijuice_gui.py depends on Tk as the graphics toolkit. However Tk is not supported under wayland, only under X11.
For wayland pijuice_gui has to be completely rewritten using a different graphics toolkit supported by wayland, e.g. Pyqt5.
Patches welcome.
Closing this issue for now.

closed 2024-01-01 by tvoverbeek