PiSupply/PiJuice #1049: PiJuice on Rpi 3b+ with Raspberry Pi OS Bookworm

larsbjorndal, 2023-12-14:
With the Bookworm OS, the pijuice.service won't start, due to a timout message: "A dependency job for pijuice.service failed." In the log, the following message are related: "/dev-i2c\x2d1.device: Job dev-i2c\x2d1.device/start timed out." I also notice that on the same hardware with Bullseye running, I have the device file /dev/i2c-1. With Bookworm I instead have /dev/i2c-2. Hope this problem can be fixed. Thanks

tvoverbeek, 2023-12-16:
On Pi 3B+ /dev/i2c-2 is the videocore i2c used e.g. for HAT eeprom and controlling the camera.
It smells like the arm i2c is not enabled (should have been done by the pijuice-base post-install script).
Try to enable I2C using raspi-config in a terminal window:
sudo raspi-config -> Interface Options -> I2c -> Yes
Check then if /dev/i2c-1 exists.
If it does `i2cdetect -y 1` should detect the PiJuice at 0x14 and the PiJuice RTC at 0x68.

Are you sure the PiJuice is properly mounted on the 40-pin GPIO header?
(Not shifted by 1 row or column, have made this mistake several times)
Anyway after installing 64-bit bookworm-lite and the pijuice-base package everything works fine for me.
**Note**: in PiJuice_cli General tab make sure Power Regulator Mode is set to POWER_SOURCE_DETECTION. Otherwise the PiJuice battery will not charge when external power is applied to the Pi3B+

larsbjorndal, 2023-12-18:
[Ton van Overbeek]

Thanks a lot for your answer! You wrote:

> On Pi 3B+ /dev/i2c-2 is the videocore i2c used e.g. for HAT eeprom and
> controlling the camera.
> It smells like the arm i2c is not enabled (should have been done by the
> pijuice-base post-install script).

Actually, arm i2c was enabled.

> Try to enable I2C using raspi-config in a terminal window:
> sudo raspi-config -> Interface Options -> I2c -> Yes
> Check then if /dev/i2c-1 exists.
> If it does i2cdetect -y 1 should detect the PiJuice at 0x14 and the PiJuice RTC
> at 0x68.

I did need to enable i2c-gpio from raspi-config as well. Then I got /dev/i2c-1 and the pijuice service starts without a problem.

> Are you sure the PiJuice is properly mounted on the 40-pin GPIO header?
> (Not shifted by 1 row or column, have made this mistake several times)

Yes, it worked with the Bullseye distro.

> Anyway after installing 64-bit bookworm-lite and the pijuice-base package
> everything works fine for me.

I have 32-bit bookworm-lite. I have a Python script to print out the battery status, and that works perfectly.

> Note: in PiJuice_cli General tab make sure Power Regulator Mode is set to
> POWER_SOURCE_DETECTION. Otherwise the PiJuice battery will not charge when
> external power is applied to the Pi3B+

Actually, pijuice_cli doesn't work. If I select Status, I get the following output:

Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 2087, in <module>
    loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 287, in run
    self._run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 385, in _run
    self.event_loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 790, in run
    self._loop()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 827, in _loop
    self._watch_files[fd]()
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 416, in <lambda>
    wrapper = lambda: self.parse_input(
                      ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 515, in parse_input
    callback(processed, processed_codes)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 412, in _update
    self.process_input(keys)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 513, in process_input
    k = self._topmost_widget.keypress(self.screen_size, k)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 598, in keypress
    return self.top_w.keypress(self.top_w_size(size,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 1626, in keypress
    key = self.focus.keypress(tsize, key)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 2316, in keypress
    key = w.keypress((mc,) + size[1:], key)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/urwid/listbox.py", line 968, in keypress
    key = focus_widget.keypress((maxcol,),key)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/urwid/wimp.py", line 543, in keypress
    self._emit('click')
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 461, in _emit
    signals.emit_signal(self, name, self, *args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 265, in emit
    result |= self._call_callback(callback, user_arg, user_args, args)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 295, in _call_callback
    return bool(callback(*args_to_pass))
                ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/bin/pijuice_cli.py", line 1994, in item_chosen
    callback()
  File "/usr/bin/pijuice_cli.py", line 209, in __init__
    self.main()
  File "/usr/bin/pijuice_cli.py", line 269, in main
    status_args = self.get_status()
                  ^^^^^^^^^^^^^^^^^
  File "/usr/bin/pijuice_cli.py", line 215, in get_status
    status = pijuice.status.GetStatus().get('data', {})
             ^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'status'

If I select General from within pijuice_cli, I get `Unable to connect to device`.

i2cdetect -y 1, gives:

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         

Thanks in advance!

Lars

tvoverbeek, 2023-12-18:
Since you are running a 32-bit userland there is a problem when you are installing the pijuice-base package.
On a Pi3B+ with a 32-bit userland the linux kernel is 64-bit.
Look at the output from `ls -l /usr/bin/pijuice*`.
/usr/bin/pijuice_cli should point to /usr/bin/pijuice_cli32 and not to /usr/bin/pijuice_cli64.
Similar for /usr/bin/pijuice_gui and /usr/bin/pijuiceboot.
Here is my output on a 64-bit userland system:
```
$ ls -l --color=none /usr/bin/pij*
lrwxrwxrwx 1 root    root        22 Nov 25 21:18 /usr/bin/pijuiceboot -> /usr/bin/pijuiceboot64
-rwxr-xr-x 1 root    root     22732 Jan 29  2022 /usr/bin/pijuiceboot32
-rwxr-xr-x 1 root    root     23904 Jan 30  2022 /usr/bin/pijuiceboot64
lrwxrwxrwx 1 root    root        22 Nov 25 21:18 /usr/bin/pijuice_cli -> /usr/bin/pijuice_cli64
-rwsr-xr-x 1 pijuice pijuice   9096 Dec 17  2018 /usr/bin/pijuice_cli32
-rwsr-xr-x 1 pijuice pijuice  14872 Jan 30  2022 /usr/bin/pijuice_cli64
-rw-r--r-- 1 pijuice pijuice 101552 May 16  2021 /usr/bin/pijuice_cli.py
lrwxrwxrwx 1 root    root        22 Nov 25 21:18 /usr/bin/pijuice_gui -> /usr/bin/pijuice_gui64
-rwsr-xr-x 1 pijuice pijuice   9096 Dec 17  2018 /usr/bin/pijuice_gui32
-rwsr-xr-x 1 pijuice pijuice  14872 Jan 30  2022 /usr/bin/pijuice_gui64
-rw-r--r-- 1 pijuice pijuice 105446 May 16  2021 /usr/bin/pijuice_gui.py
-rwxr-xr-x 1 root    root     11275 Jan 30  2022 /usr/bin/pijuice_log.py
-rwxr-xr-x 1 root    root     15664 Jan 30  2022 /usr/bin/pijuice_sys.py
-rwxr-xr-x 1 root    root      6205 Jan 30  2022 /usr/bin/pijuice_tray.py
```

larsbjorndal, 2023-12-19:
[Ton]

> Since you are running a 32-bit userland there is a problem when you are
> installing the pijuice-base package.
> On a Pi3B+ with a 32-bit userland the linux kernel is 64-bit.

uname -m gives `armv7l`

> Look at the output from ls -l /usr/bin/pijuice*.
> /usr/bin/pijuice_cli should point to /usr/bin/pijuice_cli32 and not to /usr/bin
> /pijuice_cli64.
> Similar for /usr/bin/pijuice_gui and /usr/bin/pijuiceboot.

Output here is:

lrwxrwxrwx 1 root    root        22 des.   4 20:26 /usr/bin/pijuiceboot -> /usr/bin/pijuiceboot32
-rwxr-xr-x 1 root    root     22732 jan.  29  2022 /usr/bin/pijuiceboot32
-rwxr-xr-x 1 root    root     23904 jan.  30  2022 /usr/bin/pijuiceboot64
lrwxrwxrwx 1 root    root        22 des.   4 20:26 /usr/bin/pijuice_cli -> /usr/bin/pijuice_cli32
-rwsr-xr-x 1 pijuice pijuice   9096 des.  17  2018 /usr/bin/pijuice_cli32
-rwsr-xr-x 1 pijuice pijuice  14872 jan.  30  2022 /usr/bin/pijuice_cli64
-rw-r--r-- 1 pijuice pijuice 101552 mai   16  2021 /usr/bin/pijuice_cli.py
lrwxrwxrwx 1 root    root        22 des.   8 22:48 /usr/bin/pijuice_gui -> /usr/bin/pijuice_gui32
-rwsr-xr-x 1 pijuice pijuice   9096 des.  17  2018 /usr/bin/pijuice_gui32
-rwsr-xr-x 1 pijuice pijuice  14872 jan.  30  2022 /usr/bin/pijuice_gui64
-rw-r--r-- 1 pijuice pijuice 105446 mai   16  2021 /usr/bin/pijuice_gui.py
-rwxr-xr-x 1 root    root     11275 jan.  30  2022 /usr/bin/pijuice_log.py
-rwxr-xr-x 1 root    root     15664 jan.  30  2022 /usr/bin/pijuice_sys.py
-rwxr-xr-x 1 root    root      6205 jan.  30  2022 /usr/bin/pijuice_tray.py

But maybe I should use 64-bit instead of 32-bit anyway.

Thanks,
Lars