PiSupply/PiJuice #1050: Failed to execute command "/usr/bin/pijuice_gui"

BigDogsDad, 2023-12-30:
On Pi Desktop Clicking Preferences- PiJuice setting- Throws Error that states, "Failed to execute child process "/user/bin/pijuice_gui" (No such file or directory) Currently Running Pi4 with Raspbian GNU/Linux 12 (bookworm). Any help would be appreciated. Thank you.

tvoverbeek, 2023-12-30:
Are you running 32-bit version of bookworm?
If so see Issue #1000.
64-bit version of bookworm should work.
Problem is on 32-bit bookworm the linux kernel is 64-bit which upsets the 32-bit/64-bit detection in the pijuice-base and pijuice-gui installation.

BigDogsDad, 2023-12-30:
It looks like I'm 64 but when I ran another command it shows 32 so I'm guessing 32.
[image]

BigDogsDad, 2023-12-30:
I saw in your issue #1000 that you had that fellow run the ls -l 
[image]
/user/bin/pijuice* command and this is what I got for that

BigDogsDad, 2023-12-30:
Lastly I also tried running sudo rm and sudo ln. The good news is that I no longer get the error when I click on the PiJuice seettings. Then bad news is that nothing happens at that point. No error,  no preference choices, nothing.

[image]

tvoverbeek, 2023-12-31:
This is on a pi 3b+ or pi4 running 32-bit bookwormd?
On 32-bit bookworm the linux kernel is 64-bit, but tuns 32-bit applications
`uname -a` will show aarch64 (= kernel is 64-bit).
If you are using wayland (the default in bookworm) see issue #1051 for a solution.

BigDogsDad, 2024-01-02:
To confirm I'm running a Pi4. And it seems Wayland was the way forward. I now see the GUI and am able to explore all the options. Thank you VERY much for the assistance und Ich wünsche dir ein frohes neues Jahr.

tvoverbeek, 2024-01-02:
Glad it works for you now. Closing the issue.

closed 2024-01-02 by tvoverbeek