PiSupply/PiJuice #1051: Software not opening.

Samu0535, 2023-12-31:
hello everyone, i am unable to open the PiJuice-settings software, after it is installed.
[image]

can i get some advice? Thanks in advance!

tvoverbeek, 2023-12-31:
Which OS version?
If this is Raspberrt Pi OS bookworm running Wayland then the pijuice gui does not work.
It is written with Tk as graphical toolkit. Tk is not supported under Wayland, hence app does not open.
In that case open raspi-config in a terminal (`sudo raspi-config`), 
Select Advanced Options -> Wayland -> X11 Openbox Window Manager with X11 backend.
Reboot to make it effective. Then it should work.

Samu0535, 2023-12-31:
thank you for your reply.
going to raspi-config, and applying the setting worked!!
I hope you have a wonderful day!

closed 2023-12-31 by Samu0535