PiSupply/PiJuice #1056: literally nothing working although installed

theredhorseman42, 2024-01-10:
using a pi 4 model b. originally installed the gui one through terminal and the battery correctly appeared but i would receive an error "failed to execute child process "/usr/bin/pijuice_gui" (no such file or directory)" when attempting to open the settings. i have confirmed i have that file in that location. I tried reinstalling, i searched the forums and found one about an issue with a change in the isalive() function syntax but i have the up to date one, i've tried doing it on fresh installs of both 64 and 32 bit raspbian, every forum i've found for my issue has ended at some problem i don't have. i have tried reinstalling the program then re-logging, i've tried installing then rebooting, and after trying it on 32bit OS the battery display icon is no longer there and my issue with settings not opening persists.

theredhorseman42, 2024-01-13:
that's what i thought too which is why i tried both with a fresh install, if the 32 bit is actually a 64 bit kernel as was described the issue should have been fixed by running it on the 64 bit version where the kernel didn't disagree with the os but that didn't help either. that's partially why i was confused.

tvoverbeek, 2024-01-13:
WHen using the GUI the default 32-bit and 64-vit bookworm (debian 12) installs use Wayland not X11. pijuice_gui does not work under Wayland.

cjoe-51, 2024-01-26:
> WHen using the GUI the default 32-bit and 64-vit bookworm (debian 12) installs use Wayland not X11. pijuice_gui does not work under Wayland.

Does it mean that Pijuice does not support the standard installation of Bookworm for RPI?

tvoverbeek, 2024-01-26:
It means pijuice_gui does not work on a default bookworm install on RPI 4 and 5 because of Wayland. Pijuice_gui and the service work.
Go to raspi_config advanced settings and u der Wayland select x11 and open box. Then it works fine.

michaelundwd, 2024-02-06:
Have been following this thread with interest, as I have exactly the same problems in getting pijuice_gui to work under a fresh 64-bit raspbian (bookworm).  Under Wayland, using VNC, the desktop loads, but pijice_gui is not showing and does not work under preferences.  Changing to X11 means that VNC no longer connects - the drop down properties menu shows, but the screen remains black.  So, I am left running pijuice-cli under Wayland.  Doe we know if pijuice software will be updated to include running under Wayland?

tvoverbeek, 2024-02-07:
@michaelundwd How do you change to X11?
If you use raspi-config -> advanced settings to switch from wayland to X11 then you also have to reenable the VNC server (RealVNC as before, not wayvnc).
Same when  you switch from X11 to pure wayland. 
Swiching to the appropriate vnc server is **not** automatic.

The RealVNC viewer now works with both wayvnc and RealVNC servers.
Changing pijuice_gui to work with wayland is a major job.
pijuice_gui now uses Tkinter as graphical toolkit, which is not supported under wayland.
It should be rewritten using e.g. PyQt5 (Patches/PRs welcome)

michaelundwd, 2024-02-07:
@tvoverbeek 
Brilliant, that was just the help I needed.  Pijuice_gui now opens and displays its icon correctly in a VNC window on my desktop.  

Thanks also for answering my other question about updating the; it explains why I was getting Tkinter errors when trying to run under Wayland.  Afraid I cannot offer any help with PyQt5; python's not really part of my skillset.
Michael

quitmeyer, 2024-02-28:
> @michaelundwd How do you change to X11? If you use raspi-config -> advanced settings to switch from wayland to X11 then you also have to reenable the VNC server (RealVNC as before, not wayvnc). Same [...]

These tips worked for me, thanks @tvoverbeek .