PiSupply/PiJuice #1060: PiJuice Zero Led and PiJuice battery connector references

hugokernel, 2024-02-01:
The D1 led on the PiJuice Zero was ripped off :( after I took it out of the bag, one of its contacts seems destroyed, I'd like to order a new one and replace it, can you give me its reference?

Also, can you tell me the part number of the BP7X battery connector on the PiJuice?

Thank you.

tvoverbeek, 2024-02-01:
Found this (no guarantee if it is the right part no)

D1 Led "Lite-On LTST-S310F2KT" available at e.g. Mouser
BP7X batttery connector: "AVX 00-9155-004-742-006"

closed 2024-02-01 by hugokernel

hugokernel, 2024-06-04:
@tvoverbeek Hi, can I ask you about the reference of the PiJuice Zero battery connector, do you know his reference ?

hugokernel, 2024-06-04:
Thanks for the reply.

I hadn't found the reference in the documentation :pray: