PiSupply/PiJuice #1062: Pijuice service not starting - Dependency failed for pijuice.service

mroelandts, 2024-02-03:
Device: Pi zero 2 + PiJuice zero
Firmware: V1.6
OS: 2023-12-11-raspios-bookworm-arm64-lite

After a bootup the pijuice.service is not running.
```
> sudo systemctl status pijuice.service
○ pijuice.service - PiJuice status service
     Loaded: loaded (/lib/systemd/system/pijuice.service; enabled; preset: enabled)
     Active: inactive (dead)
Dependency failed for pijuice.service - PiJuice status service.
pijuice.service: Job pijuice.service/start failed with result 'dependency'.
```
After I check what the dependency was of the service
```
> sudo systemctl status dev-i2c\\x2d1.device
○ dev-i2c\x2d1.device - /dev/i2c-1
     Loaded: loaded
     Active: inactive (dead)
dev-i2c\x2d1.device: Job dev-i2c\x2d1.device/start timed out.
Timed out waiting for device dev-i2c\x2d1.device - /dev/i2c-1.
dev-i2c\x2d1.device: Job dev-i2c\x2d1.device/start failed with result 'timeout'.
```
Why is my i2c failing on boot?
When I execute a `sudo apt install pijuice-base --reinstall` everything is back up and running!
Meaning that the "Enabling I2C" from the [postinst](https://github.com/PiSupply/PiJuice/blob/master/Software/Source/debian-base/postinst) does activate it but does not enable it for next boot.
First I though I did a bad job of soldering the header on it, but since the postinst can activate it, this might not be the case.

Any advice is much appreciated.

tvoverbeek, 2024-02-03:
Seems something is disabling i2c-1 at boot. After boot can you check with raspi-config what teh status of I2C interfacing is?

mroelandts, 2024-02-03:
I checked the I2C status in raspi-config. It was turned off ("no" was selected).
I selected yes and now everything is fully working. I can reboot without issues.
Great suggestion! Thanks for your help.

Does the pijuice-base enable the I2C interface in a different way? (and therefore not persistent across reboots)
I never had this issue before with my other pijuices. Could it be cause this is a 64bit OS?

closed 2024-02-03 by mroelandts

tvoverbeek, 2024-02-03:
pijuice-base is supposed to do the same as raspi-config. Probably due too changes in recent versions of raspi-configit does not work any more.