PiSupply/PiJuice #1063: RTC not working on PijuiceZero with Raspberry Pi OS Lite 64 bit bookworm

mroelandts, 2024-02-16:
Hello,

Device: Pi zero 2 + PiJuice zero
Firmware: V1.6
OS: 2023-12-11-raspios-bookworm-arm64-lite
Connected devices: Pijuice zero 1200mAh battery

I'm having some issues with the RTC on my latest setup.
According to the [rtc documentation](https://github.com/PiSupply/PiJuice/tree/master/Software#pijuice-rtc) the i2c-rtc module should automatically load. <br>But even while forcing it by adding `dtoverlay=i2c-rtc,ds1339` in `/boot/config.txt` does not enable it.
I do not get an ‘UU’ at address 68.

```bash
$ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --   

$ ll /dev | grep i2c
crw-rw---- 1 root i2c      89,   1 Nov 10 01:25 i2c-1
crw-rw---- 1 root i2c      89,   2 Nov 10 01:25 i2c-2

$ uname -m
aarch64
```
The EEPROM address is set to 50. (default)
[image]

The command `lsmod | grep rtc` also returns nothing, which makes me believe the module is not loaded.
Should I activate a different module for bookworm (64bit)?
I do have a working setup, but this one is running the bullseye build of raspberry OS Lite with `dtoverlay=i2c-rtc,ds1339` inside the `/boot/config.txt`.
(cat /etc/debian_version -> 11.3 && uname -mm -> armv7l)

First I had some issues with the i2c module not activate: see [PiJuice/issues/1062](https://github.com/PiSupply/PiJuice/issues/1062) but raspi-config fixed that.
Any help is much appreciated!
If more information is needed, please clarify.

tvoverbeek, 2024-02-19:
I tried to reproduce you problem using a Pi zero 2 and the Pijuice HAT.
Installing raspios 64-bit lite with the imager results in the same version as yours. After an `sudo apt update; sudo apt -y upgrade` I get version 2024-01-25.
With both the imager version and the 2024-01-25 version the pijuice-base package installs fine.
I even get the `UU` for address 68 in i2cdetect.

Have you tried manually loading the rts_ds1307 module (`sudo modprobe rtc_ds1307`)?
lsmod should ahow both rtc_1307 and regmap_i2c

tvoverbeek, 2024-02-22:
@mroelandts You wrote:
What do you mean with "the 2024-01-25 version"? Are you taking about the raspberry os or the pijuice-base package?
After running `sudo apt update` and `sudo apt -y upgrade` the output of `uname -a` is:
```
tvo@pi02-test:~ $ uname -a
Linux pi02-test 6.1.0-rpi8-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux
tvo@pi02-test:~ $ 
```

Note: You can use the Raspberry Pi Imager to download the 64-bit bookworm-lite for the Pi02 by selecting "No filtering" as device.
The problem with the PiJuice Zero is that it is not a proper HAT. It does not have an EEPROM, so the configuration (Enabling I2C and loading the rtc driver and /dev/rtc) has to be done by hand.
I'll describe how I did it tomorrow.
The end result:
```
tvo@pi02-test:~ $ sudo hwclock -rv
hwclock from util-linux 2.38.1
System Time: 1708639599.513575
Trying to open: /dev/rtc0
Using the rtc interface to the clock.
Assuming hardware clock is kept in UTC time.
Waiting for clock tick...
ioctl(4, RTC_UIE_ON, 0): Invalid argument
Waiting in loop for time from /dev/rtc0 to change
...got clock tick
Time read from Hardware Clock: 2024/02/22 22:06:40
Hw clock time : 2024/02/22 22:06:40 = 1708639600 seconds since 1969
Time since last adjustment is 1708639600 seconds
Calculated Hardware Clock drift is 0.000000 seconds
2024-02-22 23:06:39.507732+01:00
tvo@pi02-test:~ $ 
```

tvoverbeek, 2024-02-23:
@mroelandts Here the continuation of yesterday.
Same hardware configuration as you: Pi02W, PiJuice Zero with 1200mAh battery.
Installed Raspberry Pi Os Lite with the Imager (Customizing User, Password, Wifi, Localization and enable SSH)
After first run `sudo apt update` and `sudo apt -y upgrade`
Enabled I2C via raspi-config.
installed i2c-tools (for i2cdetect)
With PiJuicd Zero + battery connected, i2cdetect gives a response for addresses 14 and 68
Load the i2c-rtc device tree overlay dynamically: `sudo dtoverlay i2c-rtc ds1307`
Check that /dev/rtc exists: `ls -l /dev/rtc*`
Running i2cdetect now give `'UU'` for address 68.
Also `sudo hwclock -rv` now works. (See previous post).
After a reboot the overlay is gone and you do not get the rtc
Adding the line `dtoverlay=i2c-rtc,ds1307` to `/boot/firmware/config,txt` and an other reboot makes it permanent.

weasyx, 2024-09-04:
I'm having huge issues with PJ on Bookworm. I will open a separate issue for that. It is sad that PiSupply has not made any updates in what feels like ages. 
Regarding the config.txt file to add the RTC declaration, it has been moved to /boot/firmware/config.txt in the Bookworm release, so you need to make the modifications there and not in /boot/config.txt as it is the case in Bullseye. 
Generally I suggest you use the Bullseye version, because PJ has way too many issues in Bookworm as from my testing.
Good luck

tvoverbeek, 2024-09-04:
@weasyx I suppose you have problems with user functions under bookworm.
In bookworm home directory protectiond is set to rws------ (-> no group access to executable scripts in your home directory)
See mu comment hereL https://github.com/PiSupply/PiJuice/discussions/1080#discussioncomment-9557406