PiSupply/PiJuice #1065: Way to use a physical trigger to Disable Wakeup?

quitmeyer, 2024-03-02:
Hi! I am developing the [Mothbox](https://forum.openhardware.science/c/projects/mothbox/73) (a fun open source low cost science tool for observing nocturnal insects!)

And we have about 20 of these things we have built are are deploying them out to natural areas (with no internet connections or nothing). 

They are set to run a nightly routine when the field assistants set them up, but we don't want them to run while we are driving them to places  or shipping them or whatever. So basically we would love a **way to physically "arm" or "disarm" the devices.**

In my mind the simplest thing could be something like a jumper wire that connects GND and a GPIO pin, and if that connection is made, the device will stay inert, but when the field assistant simply takes the wire out, the device is now "armed" and will do its automatic routine (until that jumper wire might get put in place in the future).

But I'm not sure how to program the Pijuice to NOT wake up? Like I can think of scripts that while the device is running could check a GPIO pin and decide to do something on the pi, but I'm not sure of what would be the best way to put the Pi with the PIjuice into two different states like this.

Any suggestions?
Thanks for your help!

tvoverbeek, 2024-03-04:
Question: Are the boxes continuously powered for normal operations (in the armed state) or are they using the PiJuice battery?
For transport (disarmed state) the devices are unpowered (except the PiJuice could wake them up).
Correct?

quitmeyer, 2024-03-04:
The way we power them in general is we have a big battery going to the pijuice via the pijuice usb.

No other power is going to the pi itself. So the pijuice always has its battery installed and sometimes the pijuice battery is in charging state.

To make things a bit more complicated, there is a relay hat with other wires installed on top of the pijuice (hence we can't take its battery out easily). That's why I'm thinking something with the gpio?

quitmeyer, 2024-03-04:
Is it perhaps possible with

https://github.com/PiSupply/PiJuice/tree/master/Software

Digital pin io2

"IO2 can be set as a digital wake-up interrupt on the Raspberry Pi from a peripheral device connected to header P3. The wakeup can be set as one of the following:

NO_WAKEUP - Wakeup function disabled"

Would that actually disable the wake up?

[image]

quitmeyer, 2024-03-28:
I had to make a kind of hack, workaround right now.
If you look at my new "Scheduler" script that runs everytime my pi starts up, i have it check for 2 GPIO pins letting me set three states:
https://github.com/Digital-Naturalism-Laboratories/Mothbox/blob/main/Software/Scheduler.py
"ALWAYS OFF"
"DEBUG"
or
"ARMED"

it doesn't actually stay always off, it wakes up briefly and then turns itself off if it sees the GPIO pin is on OFF mode