PiSupply/PiJuice #1068: No Charging/power from USB-C/GPIO

Der-Peti, 2024-03-11:
Hello!

Device: Pi4B 4GB
PiJuice Firmware: 1.6
OS: Raspbian GNU/Linux 10 (buster) from official RetroPie image
Architecture: armhf

My problem is, tht I cannot get the Pijuice to charge from USB-C/GPIO and cannot even get the Pi powered from USB-C even if it is connected.
As soon as I disconnect the Micro USB from the Pijuice HAT, the Pi is running from the Pijuice power, even if a USB-C cable is connected to the Pi itself.
If I connect a Micro USB cable to the Pijuice, the Pi runs fine but fro battery power.

I have researched here/in the discussions/google, but nobody seems to have my problem.

In the pijuice_cli, I can see this:
[image]

No matter, waht I plug in, GPIO power input remains "NOT_PRESENT" and the positive mA show me that the Pijuice is powering the Pi.

Input precedence is set to GPIO_5V:
[image]

Am i doing something wrong, or is there something wrong with my Pi?

Thanks in advance, 
Peti

RichardVine, 2024-03-20:
I can't directly help I'm afraid but I've a similar issue with a Pijuice that I've just purchased and also can't find any related issues. I'm still researching.

Starting from everything fully powered down, it will boot with power from the USB-C connection. The PiJuice CLI shows GPIO power is being received and the battery charges fine. If I remove the USB-C power, the Pi continues to run off battery. So far, so good. However, If I restore USB-C power, the PiJuice doesn't seem to notice and continues running off battery until it eventually gives up, or I restore power via the PiJuice's micro-USB port.

Removing the battery entirely and restoring USB-C power also seems to work, at least until I remove the USB-C cable and the same problem occurs.

It's possible these _might_ be the same issue.

I'd prefer to use the USB-C power due to the power supply I have; I'm not sure if there are really any downsides in always using the micro-USB supply instead other than the need to use a different charger, or a USB-C to micro-USB adapter, which also works.

RichardVine, 2024-03-22:
I seem to have fixed my issue. Although the Raspberry Pi4 and official charger are both brand new, I swapped to a different charger. If I remove USB-C power, the PiJuice battery takes over seamlessly, then on restoring power the PiJuice now notices and resumes charging. That's how I was expecting it to work all along.

The Pi4 charger otherwise seems to work fine so I wasn't expecting to see an issue like this. It's actually the same charger I've been using with a micro-USB adaptor to plug directly into the PiJuice micro-USB port where it 'resumes' fine; it just doesn't resume with USB-C.

mattstocum, 2024-09-04:
I'm having the exact same issue. Everything was working perfectly for a while, now it's not. No clue why.