PiSupply/PiJuice #1075: Pijuice not charging via microUSB port - Input always "weak"

quitmeyer, 2024-03-28:
I have a pijuice that I cannot seem to charge the stock onboard battery. I think it had been working ok, but recently it was having what appeared to be brownouts and would be difficult to VNC into. I noticed that in the pijuice settings the USB micro input always said "Weak"

normally we have a large USB talentcell batterypack with a dedicated on-off switch that we have never had any problems with. It connects to the pijuice USB micro port and keeps the little pijuice topped up

This is a photo when just the USBC power is connected to the PI4
[image]

This is a photo when the USB C of the pi4 is connected to power AND the USB micro is connected to power
[image]

When just the Micro USB is connected and no power is going directly to the rpi4 USB C, the whole thing just instantly shuts off.

I swapped the battery with a different Pijuice in case that was a problem, but same thing. and I also tried connecting the  USB Micro to different power sources including several different chargers of 2amps or more, and it still says "weak"

Here's all my configuration options. The only thing i personally changed was ID eeprom address to 50
[image]

[image]
i noticed that the battery tab was default set to something that wasn't the standard motorola 1820mah battery, but i didn't change that myself at all

Also side question: is there a difference between connecting the extra power to my setup to the USBC of the Pi4 or the USB micro of the Pijuice? Is it also suposed to just charge via the GPIO if the power is coming from the USBC of the pi4?

tvoverbeek, 2024-03-28:
What is the real battery connected to the PiJuice? The default BP7X or really the 12000 mAh battery via screw terminal?

quitmeyer, 2024-03-28:
> What is the real battery connected to the PiJuice? The default BP7X or really the 12000 mAh battery via screw terminal?

The actual battery currently in the device is the little 1820mah that comes with the pijuice

tvoverbeek, 2024-03-28:
then it seems the dip switches have accidentally been changed from [ON] [OFF] (BP7X_1820) to [ON][ON] (PJLIPO_12000).

quitmeyer, 2024-03-28:
Hmmm, im unable to see if the dip switches have been moved at all right now (it's connected to the pi in a larger device) but i definetley didn't change any of that on purpose.

i did however switch my USB cable that was going into the microUSB port. It had been working for 2 months totally normally, but then this past week the troubles i mentioned started happening, but the micro USB cable seems to be the culprit. 

It still says "Weak" power delivery, all the time, but it seems to charge ok now.

weird to me that a USB cable that was plugged in and not moved somehow partially failed, but the quality control of USB cables in the variety of them in the world is astonishingly bad!

closed 2024-03-28 by quitmeyer