PiSupply/PiJuice #1082: Does the PiJuice work without a connected battery?

hugokernel, 2024-05-29:
Hi,

I'm using a PiJuice Zero with a 600mAh battery to power a daughter board containing a Compute Module 4 and a few peripherals (the battery is there to properly shut down the module if the main power supply is lost, hence the low capacity).

The daughterboard and Compute Module are powered by the Pijuice Zero's GPIO port.

I recently noticed that if the battery wasn't connected, it wouldn't start (it supplies the necessary 5V but stops plus starts again and so on, you can even hear a clicking noise), as soon as I connect the battery, everything works properly.

So my question is: is this normal, or does it have to be a connected battery?

Thx

tvoverbeek, 2024-05-29:
In the HAT configuration enable 'No battery turn on'
[image]

hugokernel, 2024-05-30:
You've indirectly answered my question about using the PiJuice without a battery, however, I'm unable to operate my module if a battery isn't connected and I'm well below the 2.5A (<0.7A).

Notes:
* MicroUSB current limitation is set to 2.5A.
* I have several of these modules with as many PiJuice Zeros and the behavior is the same on all of them.

So I'll try to rephrase my question a little more precisely: Is the current limitation (2.5A) the same with or without a battery?

Is the output voltage path the same whether the battery is present or not?
Do you have any ideas?

Thx

hugokernel, 2024-06-03:
I made some measurements and here are the results. My conclusion is that it's not possible to achieve a 2.5A load without a battery without supplying 10V as input.

##### Without battery

VIN 5V, max load: ~0.82A
VIN 5.5V, max load: ~1.34A
VIN 6.5V, max load: ~1.82A
VIN 7.5V, max load: ~2.15A
VIN 10V, max load: >2.5A

##### With battery
VIN 5V, max load ~2A
VIN 10V, max load >2.5A

The BQ24160 datasheet states that it's possible to achieve 2.5A at 10V and 1.5A at 6.5V, so my measurements are more or less consistent.

However, why it can't deliver a higher current without the presence of a battery escapes me a little...

Unless I've missed something in the documentation, it seems to me that this is a very important point that deserves to be mentioned.