PiSupply/PiJuice #1083: Heat and battery discolouration

cura1dev, 2024-06-06:
Our implementation of the Pi-Juice requires that it is enclosed in a case. We are using the Smarti Pi Touch Pro Case. We are using the Pi-Juice with a Raspberry pi 4 and 7inch Display.
After a few months of testing now we have noticed some discolouration on the battery possibly from heat from the circuit board below (See attached pic). As we are going to supply these as part of our product into a home environment, I am concerned that we may have issues with the battery in future. The temperature when querying the Pi-Juice is always around the 45 degrees Celsius
Has this been reported before and is it something we should be concerned about? 

[image]

tvoverbeek, 2024-06-06:
45 degreesbis a bit warm. It is the temperature when the charge circuit starts to lower the charge current.
How are you using the PiJuice? Just as a backup in case of loss of external power?
The combined load of the Pi4 and the 7 inch display is on the high side of what the BP7X/SNN5875A can supply.
Looking at the smartipi website I see there is am optional fan available. You might want to add a fan to keep the temperature below 45 degrees. But maybe that would increase the price of your product too much.
Anyway, operating continuously at the warm temperature limit of the battery will most likely shorten the life of the battery.

cura1dev, 2024-06-10:
Hi Ton, Thank you for the reply.
We are using it as a backup in case of external power loss. We have tested using the fan and it certainly does run cooler. Our application is to detect seizure like movement and sound. The concern with a fan is the application we have is using a very sensitive microphone to detect sounds and if the fan starts to make any noise after a period of being used, our application may start false alarming. Our current thought is to make a fly-lead and position the battery away from the board to keep the battery and board cooler.

tvoverbeek, 2024-06-11:
@cura1dev Thanks for your explanation.
I would recommend to place the battery away from the board using fly-leads.
Ensure the wires are thick enough to carry the required current.
FYI the used battery connector on the PiJuice HAT can be found at Digi-Key:
https://www.digikey.com/product-detail/en/te-connectivity-amp-connectors/2-179694-3/2-179694-3-ND/2259036
Might be useful when you need to make an external battery holder.

cura1dev, 2024-06-12:
@tvoverbeek Thanks for the recommendation.
We have looked at battery options. I have come across the 18650 which makes it a bit easier for mounting with a holder.
Does this battery need to have built in protection or will with PiJuice Hat circuitry handle the protection?
They obviously do not have the NTC connection. Is this an issue with using externally connected batteries?
Thank you for all you help so far.