PiSupply/PiJuice #1085: NO POWER EVENT  Will Not Trigger Script

lindelux, 2024-06-21:
In the command line pijuice I have the "No Power" event selected to trigger USER_FUNC2. USER_FUNC2 is set as "/home/pi/Scripts/red.py".  If the power is removed from the pijuice then the D2_LED should illuminate red. If I run the script "python3 /home/pi/Scripts/red.py" the LEnanoD illuminates correctly. LED settings D2 is set to USER_LED. Permission for red.py are set to "-rwxr-xr-x  1 pi pi 1121 Jun 20 18:49 red.py". The first line in the script is "#!/usr/bin/python3".

Any ideas what I am doing wrong? Thanks!

marceldbo, 2025-03-26:
@lindelux, have you been able to solve this issue? I have the same problem with two pijuice hats not wanting to execute USER_FUNC scripts on 64-bit Bookworm. I have another one on 32-bit (Bullseye), and that works fine. There is a workaround, using PiJuice-MQTT and Node-Red, but I would like to be able to use the scripts directly. I have also checked that user "pi" is part of "pijuice", the script is executable, etc, but without success. My scripts do work when running directly from the command line eg. ./no_power.

tvoverbeek, 2025-03-31:
@marceldbo @lindelux I am guessing you are hit by the new default protection in bookworm of the home directory.
It used to be the group at least had r-x (read and execute) protection, hence making the user part of the pijuice group
made executing user functions located in the home directory possible.
Under bookworm the default home direction protection is 'rwx------' and executing 
user functions owned by the home directory owner do not work.

Check your home directory protection settings and change it to 'rwxr-x---'.
Then executing user functions should work again.

marceldbo, 2025-03-31:
Hi Ton, thanks for replying and i can confirm that it works now. Changing the "pi" home directory from 'drwx------' to 'drwxr-x-r-x' using 'chmod 755 pi' did the trick :-)