PiSupply/PiJuice #1089: Urgent: WSDA-200 USB Communcation Error

matzefit, 2024-07-25:
I encountered an error in the midst of final preparations for the deployment of a remotely operated raspberry pi outfitted with a WSDA-200 USB Stick to wirelessly connect to a G Link 200 8G.
Somehow the USB Stick isn't recognized anymore. 

The exact issue is that there is a communication error when running the below code: 

`# Test connection and read firmware version with retries`
`def test_basestation_connection(com_port, retries=10, delay=5):`
`    for attempt in range(retries):`
`       try:`
`            # Establish connection to the BaseStation`
`           connection = mscl.Connection.Serial(com_port, 3000000)`
`          baseStation = mscl.BaseStation(connection)`
`            print(f"Successfully connected to BaseStation on {com_port}")`
`        `
`            # Check firmware version to ensure communication`
`            firmware_version = str(baseStation.firmwareVersion())`
`            print(f"BaseStation Firmware Version: {firmware_version}")`
`            return baseStation`
`        except mscl.Error_Communication as e:`
`            print(f"Communication error on attempt {attempt + 1}/{retries}: {e}")`
`        except Exception as e:`
`            print(f"Error connecting to BaseStation on {com_port} on attempt {attempt + 1}/{retries}: {e}")`
`        time.sleep(delay)`
`    return None`
`com_port = "/dev/ttyUSB0"`
`baseStation = test_basestation_connection(com_port)`
`if baseStation is None:`
`   print("Failed to connect to BaseStation after multiple attempts. Exiting...")`
`   sys.exit(1)  # Exit the script if connection fails`

The output of above script is the following: 
_Successfully connected to BaseStation on /dev/ttyUSB0
Communication error on attempt 1/10: Failed to read the Firmware Version (EEPROM 108) from the BaseStation
Successfully connected to BaseStation on /dev/ttyUSB0
Communication error on attempt 2/10: Failed to read the Firmware Version (EEPROM 108) from the BaseStation
Successfully connected to BaseStation on /dev/ttyUSB0
Communication error on attempt 3/10: Failed to read the Firmware Version (EEPROM 108) from the BaseStation
Successfully connected to BaseStation on /dev/ttyUSB0
... and so on until 10 retries are done, then: 
Failed to connect to BaseStation after multiple attempts. Exiting...._

Basically when trying to get the firmware, the exception comes into effect and outputs the error that the communication cannot be established and the firmware cannot be read from the device. 

This code worked perfectly fine in the past. Could you point me to a solution on this issue? I tried rebooting the system and I tested the connection with different WSDA-200USB sticks, all of which eventually end up with this error.

shawaj, 2024-07-25:
I guess you posted this on the wrong GitHub as this WSDA-200USB is not made by us

matzefit, 2024-07-25:
oh im sorry, just noticed. I will remove this.

closed 2024-07-25 by matzefit

shawaj, 2024-07-25:
Just as an idea, one thing to check is that your RasPi has good power supply. You can get weird communications issues on USB and other communication busses if the device doesn't have sufficient power