PiSupply/PiJuice #1091: LIPO Battery 12000 mAh drains incredibly fast

vmodamio, 2024-08-21:
After optimizing my radxa-rock board (cortex A-35) for minimal power consumption, I managed to get it working around 40 mA. The pijuice status reports values oscillating between 0.020 and 0.140 A. I'm using the pijuice LIPO12000 mAh, so I was  expecting something like 12000 mAh * ( 3.7 V/ 5 V) * 0.75 (efficiency) / 100 mA (to be realistic) = 67 hr.
But the battery goes from 80% to 20% in barely 6hr!! What could it be wrong?
The battery profile was updated into the battery before using this board, when it was connected to a raspberry Pi. Even now in the new system I can set the profile again, but nothing changes.
The PiJuice is the 'raspberry-zero-like' format. It is charged via J4 (power) with cables soldered to a usb plug, connected to a 5V power supply. 
I noticed it charges very quick (+50% in less than 1hr) and the temperature does not exceed 29 degree (NTC), so I was wondering if the PiJuice is actually recognizing it as another battery?
Im using LDO regulator. Also, the Rsoc is estimated DIRECT_BY_MCU. Even though none of those configurations change anything.
Also, the battery has been charged only 5-7 cycles, and never more than 85% (never discharged below 20% neither).
What can be wrong??

zeitkunst, 2024-09-10:
I've also found extremely poor performance with the 12000 mAh battery. I haven't done super detailed characterization of it yet, but somehow it is being drained even though my PiJuice is connected to a 12V AGM battery with solar panels (that itself is being charged by a Victron charge controller). My Pi Zero runs around 0,1-0,2 A. Even though the PiJuice has a continual source of power (that is, the AGM battery never goes below 50%), my 12000 mAh battery on the PiJuice still drains rather quickly...i.e., it goes from full to 40% in two days (even though it shouldn't be drawing any load from it). I haven't been able to figure out what is going on. And yes, I've adjusted the battery profile properly, it's the PiJuice branded battery, etc.

vmodamio, 2024-09-12:
I had to call SetPowerOff(delay) function (see the API documentation) to remove the power to the board before halt. This made a big difference.

zeitkunst, 2024-09-12:
Thanks, I already use SetPowerOff in my shutdown routine, so there is something else going on (and I'm hoping that someone knows the solutions to our issues, if they're actually related!).