PiSupply/PiJuice #1094: Pi Juice question

TonyPdepository, 2024-09-08:
The very basic part of charging and powering works on a PI5 but not getting the battery level etc via python code no because the last known code uses RPI.GPIO. The software online is 4 plus years old and PI5 RPI.GPIO does not work with it and the code is based on it.
It needs to be updated to use gpiozero or coding that works on PI5 with out breaking anything. Can you or anyone help?

tvoverbeek, 2024-09-08:
@TonyPdepository Your comment about RPi.GPIO and Pi5 is correct.
zhowever the PiJuice Hat and PiJuice Zero do not use Rpi.GPIO.
All communication is done through I2C.
Works fine with a Pi5.
When you look at the source code (https://github.com/PiSupply/PiJuice/blob/master/Software/Source/pijuice.py) you will find no reference to RPi.GPIO.

TonyPdepository, 2024-09-09:
I have everything installed and I know it does not should up as it does in the picture. I know the software does not support wayland. Can you then provide python code to read the values then? Seems all I find is reference to RPi.GPIO. Thanks..

Omnipresent77, 2024-11-11:
So After all I did in https://github.com/PiSupply/PiJuice/issues/1101
I am now to this same situation.
I went into config.txt and have turned on i2c_arm, 12s, and spi, and still no icon in the tray.

running raspian 5.4 bookworm october 22, 2024, and pijuice firmware 1.6

Omnipresent77, 2024-11-12:
Okay more digging As of https://github.com/PiSupply/PiJuice/issues/1040#issuecomment-1852673293
I am on the new type of what is it lawb? wayland. when changing to X11 i can not get on reboot for it to go to a desktop even after installing one. So my only working desktop right now is a wayland one. i DID get the GUI to open, but not close. So as to that yes I guess the gui and the tray files need to some how get a wayland versions. shrugs