PiSupply/PiJuice #1096: External RTC module?

zeitkunst, 2024-09-11:
I was wondering how best to use an external RTC module (such as https://www.kiwi-electronics.com/en/adafruit-pirtc-precise-ds3231-real-time-clock-for-raspberry-pi-4313 ) in conjunction with the PiJuice and my Pi Zero 2 W. 

Here are the details of the installation: it's in an off-grid location where there is not internet access (or, there is only Internet access when I am physically present and connected to my Pi Zero with my cellular hotspot). I am using a solar panel (through a Victron charge controller) to power a 14Ah AGM battery which then powers the PiJuice. I also have the 12.000mAh battery connected to the PiJuice. The Pi Zero draws around 0,1 to 0,2 A, depending on load. The application is a long-duration time-lapse which needs to turn on at sunrise and turn off at sunset. The Pi Zero calculates these times based on the date. As a result, having accurate, continual time is necessary.

Because of the vagaries of the sun in the install location (Amsterdam), I've had issues with the PiJuice running out of LiPo battery, causing the time to be lost. This is because I have logic to turn on and off the LiPo battery charging on the PiJuice so as to not drain the AGM completely overnight (which nearly happened recently).

In any event, I would like a more stable source of time that can survive the (potential) exhaustion of the LiPo, namely an external, hardware, coin-cell-backup RTC. (I trust a coin-cell-backed RTC much more than a LiPo one.) My main question is how to integrate this RTC into my setup so that the PiJuice draws its time from that, rather than its own emulated RTC? Do I just need to set a different I2C address, namely the one of the external RTC board? Or is there other configuration needed?

Thanks in advance!

tvoverbeek, 2024-09-18:
You can connect an external RTC. You end up with two RTCs in the system: the DS3231 and the emulated RTC in the PiJuice HAT.
Both on address 0x68, which is no good.
Change the emulated RTC address to something else in the PiJuice HAT Config settings to avoid the I2C conflict.

If you want to use PiJuice functions like alarms or RTC wakeups you need to copy the time from the hardware RTC to the emulated RTC either at boot or regularly with cron.

zeitkunst, 2024-09-19:
Okay, thanks for the information. So this still requires the Pi to powerup in order to provide the date in the external RTC to the PiJuice RTC. Thus having the external RTC would not help in the case of the PiJuice losing power and thus being unable to respond to a wakeup alarm. (I cannot use the PiJuice wakeup on charge feature because I turn off the PiJuice charging from the Pi before I shutdown the installation at night, so as to avoid potentially draining my AGM batteries overnight in the case that the PiJuice needs to charge its battery overnight.) And I just looked at the datasheet for the STM32 used in the PiJuice and there does not appear to be an option to use an external battery to power the STM32 RTC like there is on some other STM32 processors.

I guess from all of this there's no way to ensure that the STM32 RTC retains its date (and thus ability to respond to a wakeup alarm) across the loss of both battery power to the PiJuice and loss of power to the PiJuice over USB. The reason why this is important to me is if the Pi does not wakeup due to the wakeup alarm then I have to physically go to the installation to turn it back on. My system would be much more resilient if I could ensure a coin-cell backup to the RTC to at least guarantee that the RTC retains the date across potential draining of the PiJuice LiPo and low battery conditions on the AGM...which both can be remedied by waiting for the AGM to get charged back up by the solar panel, thus eventually allowing the PiJuice LiPo to be charged up again once the RTC tells the PiJuice (and thus the Pi) to turn back on and start charging the LiPo again.

Would it be potentially possible to modify the PiJuice firmware to use an external RTC rather than its own?