PiSupply/PiJuice #1100: COMMUNICATION_ERROR I2C address 0x08

dpezto, 2024-10-10:
I just received a 2nd hand PiJuice HAT and I'm having trouble setting it up. When running the CLI the HAT status shows `Fault: COMMUNICATION_ERROR`.
I ran the `i2cdetect -y 1` and got:
```bash
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         08 -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 
```
I tried the [PiJuice bricked after unsuccessful firmware update](https://github.com/PiSupply/PiJuice/tree/master/Firmware#pijuice-bricked-after-unsucsessful-firmware-update) using address `0x08` and got the firmware successfully flashed but still `pijuice_cli` shows the communication error.

What am I missing here?

tvoverbeek, 2024-10-10:
Is the `i2cdetect -y 1` output after a reboot still the same?

dpezto, 2024-10-13:
Yes, it's still the same.

tvoverbeek, 2024-10-13:
It seems the previous owner changed the I2C address from the default 14 to 8.
pijuice_cli by default uses address 0x14, since it does not know the address has changed.
It can be changed in the CLI or GUI and the non-standard address is stored in the pijuice settings file `/var/lib/pijuice/pijuice_config.JSON`.
Edit the settings file `/var/lib/pijuice/pijuice_config.JSON` by adding the following lines before the closing brace:
```
  "board": {
    "general": {
      "i2c_addr": "8"
    }
  }
```
Note 1: Each line starts with 2 spaces
Note 2: You have to be root in order to edit the file.

Try this and see if you can run pijuice_cli without communication error.
In the General tab you can then reset the I2C adddress to its default.

dpezto, 2024-10-13:
That was it, thank you!

closed 2024-10-13 by dpezto