PiSupply/PiJuice #1104: PiJuice 22W Solar Panel merging outputs

gabrieltakacs, 2024-11-20:
I've bought PiJuice 22W Solar Panel for my single Raspberry PI Zero with Pijuice Hat. I've then realized that my solar panel has two outputs instead of one. Can I merge those two USB outputs to one and provide full power for my single Raspberry device?

tvoverbeek, 2024-11-22:
You can merge the two outputs to get a max output of 20 Watts.
It does not make sense to expect the PiJuice HAT to just pass through this max power (i.e 5V,4A) to the PiZero.
The conversions in the PiJuice only allow a max current to the Pi of 2.5A (See https://github.com/PiSupply/PiJuice/tree/master/Hardware#power-management-and-batteries)

gabrieltakacs, 2024-11-25:
I understand that. The only reason why I would like to merge two outputs is that in winter months when my panels can produce only very limited energy for a short time, for example I don't get full 12W but only 3-5W from one output. I think if I merge outputs, I might get for example 2x400mA = 800mA instead of only 400mA. Am I right?

tvoverbeek, 2024-11-25:
Yes, that should be the case (although I never have tested this because I do not have the 22W solar panel).