PiSupply/PiJuice #1105: PiJuice not working or communicating

jbmorris289, 2024-11-27:
Hi. My Pi has been having trouble communicating with my PiJuice. I cannot do much of anything with it. This is the error I get from pijuice_cli:
```
Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 2087, in <module>
    loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 287, in run
    self._run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 385, in _run
    self.event_loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 790, in run
    self._loop()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 827, in _loop
    self._watch_files[fd]()
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 416, in <lambda>
    wrapper = lambda: self.parse_input(
                      ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 515, in parse_input
    callback(processed, processed_codes)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 412, in _update
    self.process_input(keys)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 513, in process_input
    k = self._topmost_widget.keypress(self.screen_size, k)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 598, in keypress
    return self.top_w.keypress(self.top_w_size(size,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 1626, in keypress
    key = self.focus.keypress(tsize, key)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 2316, in keypress
    key = w.keypress((mc,) + size[1:], key)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/urwid/listbox.py", line 968, in keypress
    key = focus_widget.keypress((maxcol,),key)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/urwid/wimp.py", line 543, in keypress
    self._emit('click')
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 461, in _emit
    signals.emit_signal(self, name, self, *args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 265, in emit
    result |= self._call_callback(callback, user_arg, user_args, args)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 295, in _call_callback
    return bool(callback(*args_to_pass))
                ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/bin/pijuice_cli.py", line 1994, in item_chosen
    callback()
  File "/usr/bin/pijuice_cli.py", line 209, in __init__
    self.main()
  File "/usr/bin/pijuice_cli.py", line 269, in main
    status_args = self.get_status()
                  ^^^^^^^^^^^^^^^^^
  File "/usr/bin/pijuice_cli.py", line 215, in get_status
    status = pijuice.status.GetStatus().get('data', {})
             ^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'status'
```

tvoverbeek, 2024-11-27:
This smells like I2C is not initialized and/or the PiJuice has a changed I2C address.
Can you check the output from `i2cdetect -y 1`?
Which Pi and which OS? (output from `cat/etc/debian_version` and `uname -a`)

jbmorris289, 2024-11-27:
> This smells like I2C is not initialized and/or the PiJuice has a changed I2C address. Can you check the output from `i2cdetect -y 1`? Which Pi and which OS? (output from `cat/etc/debian_version` and `uname -a`)

That would be interesting, because it was working before...

`i2cdetect -y 1` returns:

> Error: Could not open file \`/dev/i2c-1' or `/dev/i2c/1': No such file or directory

I have a Pi 4 Model B, 2GB. 
This was the output from those two commands:

> 12.8
> Linux [hostname redacted] 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

jbmorris289, 2024-11-29:
> You are obviously on Bookworm (Debian 12.8). Has it ever worked after you upgraded to Bookworm? Now config.txt is located in /boot/firmware. The file /boot/config.txt now contains:
> 
> ```
> DO NOT EDIT THIS FILE
> 
> The file you are looking for has moved to /boot/firmware/config.txt
> ```
> 
> Suggest to run `sudo raspi-config` and under Interfaces enable I2C. After a reboot /dev/i2c-1 should show up and `i2cdetect -y 1` should work.

Yep, bookworm. Now I don't remember if it worked after I upgraded it.

That did it. Guess the setting is off by default - I did a headless install. Results:

```:~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

jbmorris289, 2025-03-16:
Hello,

So my PiJuice still hadn't been working for a while. Finally got to it again today - turns out it was because the battery was totally discharged. So I turned off the Pi and plugged my adapter into the PiJuice's USB port, and that solved the issue. Plugged it back into the Pi, and things work as normal now. Just a future reference to anyone who's i2c settings and others are correct but there is no PiJuice detection - try plugging into the PiJuice, or charging the battery itself if you have the tools to do that.

Pi Info: Pi 4 Model B (2GB).

closed 2025-03-16 by jbmorris289