PiSupply/PiJuice #1106: PiJuice hat GUI not working with Pi5

EnnPeeTee, 2024-11-30:
I am trying to get a PiJuice hat GUI working with a Pi5.

After installing pijuice_gui , the standard icon does not appear in the taskbar, and whilst PiJuice Settings appears in the Preferences menu, it produces the warning;

_Failed to execute child process "_/usr/bin/pijuice_gui_" (No such file or directory)._

Inspecting /usr/bin/ reveals that a multiplicity of pijuce_xxxx files are present. Is there a simple solution?

Incidentally, there is a further warning that pops up at boot telling me the PiJuice "... is not capable of supplying 5A
Power to peripherals will be restricted" - this is rather academic for my application only has a GPS device attached which consumes a mere 85mA from the USB bus. Can I disregard this warning or is there a method of telling Pi5 to downgrade the available power on the collective USB bus?

Any guidance or pointing in the right direction will be greatly appreciated!

tvoverbeek, 2024-11-30:
The default OS installation for the Pi5 uses Wayland for display (not X11).
pijuice_gui uses the "old" TkInter (TCL/Tk) for the displsy of the GUI.
TkInter is *not* supported under Wayland.
So for the Pi5 you can only use pijuice_cli in a terminal window.

If you really want to use  the GUI you can run raspi-config (`sudo raspi-config` in a terminal),
Select. `Advanced Options` -> `Wayland` -> `X11` -> `OK`
After a reboot pijuice_gui should work.
With a proper install of pijuice the output of `ls -l /usr/bin/pijuice_gui` should be:
```
$ ls -l --color=none /usr/bin/pijuice_gui
lrwxrwxrwx 1 root root 22 Jan  7  2024 /usr/bin/pijuice_gui -> /usr/bin/pijuice_gui64
```
To get the GUI working on Wayland it has to be completely rewritten with a graohics toolkit supported by Wayland (i.e PyQt5)

Regarding the message `the PiJuice "... is not capable of supplying 5A Power to peripherals will be restricted" `
The message is correct since the Pijuice can only deliver about 2A at 5V. For normal use (e.g. your academic case with only a GPS device) it should work without problems. Trying to use power hungry peripherals will be problematic)

EnnPeeTee, 2024-11-30:
Thank you so much for your excellent and most expansive guidance - it is greatly appreciated!

I shall have to look at command line operation, but I am new to the Pi and as a Windows person, [forgive me - it seemed a good idea at the time!] this means a new learning curve, but I coped with CP/M & DOS. It is just the brain is 50 years older.......  Ha!

Many thanks again!

tvoverbeek, 2024-11-30:
@EnnPeeTee Just FYI pijuice_cli is a menu driven text based program. if you run it in. a terminal window on the desktop, you can even use mouse clicks to make a selection (no need to use the direction keys to get to the selection you want).

PS I also know what CP/M is. Also used Atari TOS which was derived from CP/M.

EnnPeeTee, 2024-12-01:
After much fiddling trying to get pijuice_cli to run, I eventually replaced the 'official' OS as supplied on a spare SD card, with a fresh installation and it worked first time after installation.

However the following simple script which runs nicely on 3b+ models does not appear to respond when power is pulled, despite the PiJuice being set on 'No power:' to call shutdown.py which comprises;

#!/usr/bin/python3

from pijuice import PiJuice
import os
pijuice = PiJuice(1, 0x14)

# Remove power to PiJuice MCU IO pins
pijuice.power.SetSystemPowerSwitch(0)

# Set wakeup
pijuice.power.SetWakeUpOnCharge(0.0)

# Remove 5V power to RPi after 5 seconds
pijuice.power.SetPowerOff(5)

# Shut down the RPi
os.system("sudo halt")

The file has been made executable and ownership changed to Pi

_-rwxr-xr-x 1 pi pi  329 Dec  1 13:34 shutdown.py_

Am I missing a trick here?

tvoverbeek, 2024-12-01:
User scripts are executed by the pijuice service which runs as user pijuice.
When you do an install of Bookworm the default user directory gets the
protection `rwx------`.
This ensures only the default user can execute scripts located in the $HOME
directory.
Before Bookworm the default user directory got the protection `rwxr-x----`
meaning all users
belonging to the same groups  as the default user can execute scripts in
$HOME.
The pijuice install adds the default user to the pijuice group.

To get your shutdown.py script to execute change the protection of $HOME:
`chmod 750 $HOME` Then it should work.

EnnPeeTee, 2024-12-02:
Thanks for the info.     The CLI variant will be perfectly adequate for my needs, if I ever [ever] get the device to shut down properly.

After yet another whole day stumbling through endless dead-ends I decided to perform a completely clean installation and start again. Not that I am aware it makes any difference, all actions were performed on the Pi, as opposed to via a mixture of  SSH and VNC from a Windows machine.

'chmod 750 $HOME' stops any shutdown. In fact it prohibits Pijuice accessing /home/pi so it cannot see the shutdown.py file.  All that happened was the Pi would never shut down when the power was pulled, and it just continued to power off the battery.

Reverting to 'chmod 755 $HOME' overcomes that problem but the only controlled shutdown period I can get when power is pulled is about 5 seconds. Setting the delay to 30 second, 1 minute etc has no effect.  I therefore have no control over the duration. Incidentally I have modified the shutdown script to log the events and indeed pijuice is running the file but totally disregarding the set shutdown delay value.

Another matter which concerns me greatly is that despite a proud declaration that the Pijuice is factory-set to the correct battery as supplied, 3 of my pijuice hats were set for PJLIPO [12AH]  batteries yet the standard BP&X [1.82AH] battery is fitted.  I am concerned that overcharging will account for reduced capacity.

So I am well and truly stuck and beginning to wonder if PiJuice is compatible with Bookworm/Pi5.  Regrettably I need to use Pi5s because of their processing speed and this means unfortunately I cannot revert to Bullseye, which I have running on Pi3B+s and PiJuice hats work brilliantly and shut down exactly as expected. So frustrating! Arrgh!

EnnPeeTee, 2024-12-03:
Could this be a Python3 version issue?      The perfectly performing 3B+ units are running Python3 v 3.9.2, whereas the P5s are running Python3 v 3.11.2