PiSupply/PiJuice #1109: Will the PiJuice be sufficient for our school project?

albinhed, 2025-02-19:
Hi!

We are developing an instrumented and motorized exoskeleton for physiotherapy. We're using a Raspberry Pi Pico 2 as an MCU to collect sensor data and control the actuator. The actuator will only be used during therapy sessions at the hospital, and during these sessions, the therapist needs access to sensor data collected since the last session. The Pico will gather sensor data from daily activities and store the logs on an SD card. It will then wake the Raspberry Pi 5 at scheduled intervals to transfer the data before shutting it down again to conserve power. Since the RPi 5 is the most power-intensive component, this approach allows us to maximize battery life.

The only time the Raspberry Pi 5 will be powered continuously is during therapy sessions, which last at most 1-2 hours. During these sessions, the therapist will connect to a web server hosted on the RPi 5 to view sensor data logs since the last meeting. At the same time, the motor (powered separately) will be controlled using ROS 2 as the patient performs their therapeutic exercises.

Our concern is whether the PiJuice HAT, with its maximum output of 2.5A, will be sufficient to power the Raspberry Pi 5 while hosting the web server, running ROS 2, and handling communication for up to two hours. Would this be enough, or would it result in throttling and voltage drops?

Any insights or recommendations would be greatly appreciated!

Thanks in advance.

tvoverbeek, 2025-02-19:
If you are running on the Pi5 web server, ROS and communication (I assume WiFi) then the 2.5A max output is not a problem.
However getting 2 hours run time using the default battery (BPX1750) will be problematic.
Suggest to use a larger caoacity battery (5000mAh or 12000mAh).
Anyway, make sure you can recharge/top up the battery between therapy sessions,

albinhed, 2025-02-19:
@tvoverbeek  Thanks for your response! We’ll definitely be using a higher-capacity battery. My main concern was the 2.5A max output.

Also, from what I understand, the Pico 2 can still be powered by the PiJuice HAT even when the Raspberry Pi 5 is turned off. Additionally, the Pico 2, along with the RTC, can be used to wake up and shut down the Raspberry Pi 5 at different times.

Is my understanding correct? I’m still new to this, but I’m eager to learn!