PiSupply/PiJuice #1110: Long press 1 masking long press 2 on SW2?

Jacopo00811, 2025-02-24:
Hello there,

While reading the documentation, I didn’t fully understand whether this is the expected behavior when a user script and a system function are assigned to the same button press but have different durations. In my current configuration, they seem to be interfering with each other. Specifically:

[image]

- If the user function is triggered (a python script), the second function (SYS_FUNC_SYS_OFF_HALT) won’t run.
- If I switch them in the GUI panel, the opposite happens, the SYS_FUNC_SYS_OFF_HALT function is called preventing the user script from running.
Is this the intended behavior? If not what could be the reason of this masking? My user function is just producing a sound through a speaker and make Pi's network service becoming accessible.

tvoverbeek, 2025-02-24:
Both user functions and system functions are executed in line by the pijuice service.
If your user function is running a long time the second function has no chanve to execute.
Similarly if SYS_FUNC_SYS_OFF_HALT is already executing (shutting down the system) the 2nd function cannot execute.

Jacopo00811, 2025-02-25:
Thanks for clarifying that!

closed 2025-02-25 by Jacopo00811