PiSupply/PiJuice #1115: Start RPi when only power from battery is provided

albinhed, 2025-03-20:
Hi!

As i have understood it, i need to connect a power source to the PiJuice everytime i want to start the RPi from the SW1. 

Does that mean i can never start it by just providing power from the battery?

tvoverbeek, 2025-03-20:
No, pressing SW1 with sufficient capacity in the battery will start the. RPi.
Just tried it with a PiJuice on a Pi5 with no other power source.
Of course, if you have defined to shut down the Pi if there is no external power, then it does not work.

closed 2025-03-20 by tvoverbeek

albinhed, 2025-05-01:
@tvoverbeek Is there a setting in the Pijuice Client that does this? Because i can't seem to find it.

albinhed, 2025-05-01:
The Pi 5 initially boots when i press the button and the active cooler fans that i use starts spinning. But after a few second it's like a "hard shutdown". 

Maybe the BP7X battery that i use is too weak and the booting process draws to much current?