PiSupply/PiJuice #1117: External LED for Charging Status

islandmonkey, 2025-04-13:
I have a standalone system where I would like to add an LED to its enclosure to indicate when the PiJuice is charging, including when the Raspberry Pi is off.

- I can't use any light guides.

- As far as I understand, the free I/O pins on P3 are only usable while the Raspberry Pi is running.

- Is it possible to use the unpopulated D3 pads to add an external charging LED?

[image]

tvoverbeek, 2025-04-15:
Yes, yiu could use the D3 pads.
Tested it with PiJuice standalone (no RPi).
When charging (via the PiJuice microUSB) there is about 6.7V across the D3 pads.
When not charging, nothing (0.0V).

islandmonkey, 2025-04-15:
@tvoverbeek thanks for response!

So i could connect to the pads as follows to drive an external led? 

[image]

[image]

islandmonkey, 2025-04-15:
In case it's useful for someone else: I tested the following circuit, which worked well with an external LED. Ended up not using the STAT pad.

[image]

closed 2025-05-06 by islandmonkey