PiSupply/PiJuice #1120: User function doesn't run

GerardM55, 2025-05-14:
Hello, I’m working with the PijuiceZero and the user functions are not working for me. I’m using the SW2 which works with the rest of the actions, so the problem is not with SW2 because I’ve also set a USER_FUNC for system start and it doesn’t work either. I’ve looked at GitHub issues with similar errors and I’ve fixed my mistakes, but it still doesn’t work. I’m attaching the scripts and screenshots because I don’t know where the error might be. 

In the script ads1292.py, I want to use my virtual environment called my_env.
**inicio_script.sh**
`#!/usr/bin/env bash

# Crear el directorio de logs si no existe
mkdir -p /home/raspberry/logs

# Registro de inicio del script
echo "[$(date)] Ejecutando ads1292.py" >> /home/raspberry/logs/ejecuciones.log

# Ejecutar el script de Python en segundo plano con su intérprete
/home/raspberry/myenv/bin/python /home/raspberry/ads1292.py >> /home/raspberry/logs/ads1292.log 2>&1 &

# Registro de finalización del script bash (no del Python)
echo "[$(date)] ads1292.py finalizado (lanzado en background)" >> /home/raspberry/logs/ejecuciones.log`

**ads1292.py**
`#!/usr/bin/python3

# Abrir el archivo de log para registrar cuándo se ejecuta el script
with open("/home/raspberry/logs/ads1292_execution.log", "a") as log_file:
    log_file.write("[Iniciando ejecución de ads1292.py]\n")

def main():
    print("✅ Script corriendo... esperando eventos (Ctrl+C para salir).")
    
    # Aquí puedes poner la lógica real, por ejemplo esperar o leer datos, etc.
    
    print("⏱️ Espera de 5 segundos finalizada.")
    
    # Registrar la finalización del script en el log
    with open("/home/raspberry/logs/ads1292_execution.log", "a") as log_file:
        log_file.write("[Terminando ejecución de ads1292.py]\n")

if __name__ == "__main__":
    main()
`

Thank you

[image]
[image]
[image]
[image]

tvoverbeek, 2025-05-15:
Are you running under bookworm?
Your user scripts are in your home directory.
The scripts are executed by the pijuice service running as user pijuice.
User raspberry should belong to the pijuice group.
Check with `cat /etc/group | grep raspberry`
Under bookworm the default home directory protection changed to `rwx------`.
Hence the pijuice service fails to execute the user scripts.
Change your home directory orotection  to `rwxr-x---`.
Then it should work

GerardM55, 2025-05-15:
Now working. 
The problem was this: Change your home directory orotection to rwxr-x---.
And I also had to change the environment directory doing chmod 750 /home/raspberry/myenv. 

Thank you

closed 2025-05-15 by GerardM55